Kaldek/rat-ratgdo #100: Garage Door Opens but does not close

Stuart5851, 2024-04-16:
As per the title, I can use the webpage to open/stop but not close the garage door, any help would be appreciated.

foreverimagining, 2024-04-16:
What firmware are you running, and do you have obstruction sensors attached? Does the `Toggle` command work to close the door rather than the discrete `Close` command?

Stuart5851, 2024-04-16:
Thanks for the reply, seems like it is a known issue with certain GDO's that don't have obstruction sensors attached as they are optional. 
Yes toggle command works.  
Looks to be a couple of solutions, buy obstructions sensors or potentially wait to see if it can be bypassed. Going to close the issue as I forgot to check the main ratgdo page first and its already reported underhttps://github.com/ratgdo/esphome-ratgdo/issues/146

closed 2024-04-16 by Stuart5851

foreverimagining, 2024-04-16:
Yup, that's about the sum of it. 

It seemed that at least one person was able to make it work, but for most, it's either add obstruction sensors, or you could use a workaround like the MQTT code where you toggle the door conditionally on whether it is open when you need to specifically close it. Of course, that requires that the RATGDO or RAT-RATGDO always have accurate statuses, so it is a little sketchy sometimes.