Kaldek/rat-ratgdo #109: Only able to open and close door

dig12345, 2024-08-25:
Hi, I built the ESP32 board using [this schematic ](https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20ESP-32_schem_v6.png) and connected it to a Esp32 using the GPIO pins for ESp32 . Flashed 2.0 firmware per the readme. I am able to toggle the door open and closed which is cool but I'm not receiving any data about state of the door etc. I checked wiring far as I can tell everything is correct. I'm also unable to toggle the light or remote lock. I am getting obstruction sensor status correctly. Any ideas?

Thanks

audioeptesicus, 2024-09-13:
I just built mine too, using an ESP32 board like yourself, following the schematic and and flashing with the 2.0 firmware.

I can trigger door up, door down, the lights, but I can't stop the door or read the position.

With the door opened, pressing the close button:
`
10:15:48 | [D] | [cover:076] | 'Door' - Setting
-- | -- | -- | --
10:15:48 | [D] | [cover:084] | Position: 0%
10:15:48 | [D] | [number:012] | 'Rolling code counter': Sending state 445.000000
10:15:50 | [D] | [ratgdo:221] | Motor: state=ON
`

While the door is closing, I press the stop button and it reports that the door is not moving:
`
10:15:51 | [D] | [cover:076] | 'Door' - Setting
10:15:51 | [D] | [cover:080] | Command: STOP
10:15:51 | [W] | [ratgdo:484] | The door is not moving.
`

Any thoughts here?

dig12345, 2024-09-13:
Ended up saving my time, swallowing pride and purchasing this handsome piece:

https://www.gelidus.ca/product/gelidus-research-ac-powered-ratgdo-compatible-board/

audioeptesicus, 2024-09-13:
I had an ESP8266 lying around and tried that instead of the ESP32.  The same thing happens; I can't read the position or stop the door.  Only up and down work.

seanmcne, 2025-02-05:
> Ended up saving my time, swallowing pride and purchasing this handsome piece:
> 
> https://www.gelidus.ca/product/gelidus-research-ac-powered-ratgdo-compatible-board/

That's a solid solution and price, I can't even order PCB's and parts for that.  I like how it's all-in-one with a power supply as well.  So long as it's got true parity of the firmware, I think I'm going to give this a shot.  The only thing that would be great is if you could order it without an AC plug and could provide a 12v to 5.5v power supply for scenarios where you can run off the garage door openers 12v battery (like the 8500w).  Thanks for sharing that supplier link!