Kaldek/rat-ratgdo #128: ESP32-C3 SuperMini support

geodic, 2025-06-30:
I have recently had the time to build an ESP32-C3 SuperMIni variant of the rat-ratgdo and just wanted to create this issue to post my progress if anyone is interested. I modified the community schematics to fit the SuperMini and have created and built a breadboard design in Fritzing. The file is available here: [Google Drive](https://drive.google.com/file/d/1Ogmsm33guEwdVDG1onv_QtxVysXJzZX_/view?usp=sharing). I have flashed the v2.5 firmware through ESPHome and it seems to boot and run fairly well. However, I have not had time to mount and wire my design yet, so I cannot comment on its functionality.

geodic, 2025-07-08:
I don't think that is a modification as 
 I copied the ESP32 schematic exactly... In any case, if I'm not mistaken, that would be a voltage divider that halves the 12V from the control wire, so that only 6V is passed to the MOSFET. I guess it is technically not needed as these MOSFETs are typically rated for 20+V, but if you have the parts, I would recommend it anyway as it puts less of a strain on the MOSFET.

KnightDoom, 2025-07-08:
But that is the wrong configuration for a resistive divider. 
That diagram pulls the 12V line to ground (Via R3 10K resistor). The R4 does nothing as no current is expected to the gate.
A resistive divider would require R4 to be inline with the 12V input and Q5's gate to be tapping the midpoint.

Can you provide a reference to the ESP32 schematic you used? 
I don't see it in the ones on this repo.

geodic, 2025-07-08:
Hmm, interesting... in that case, I do not know its functionality... Here is the schematic I copied: [RatGDO-OpenSource-D1Mini-ESP32.pdf](https://github.com/Kaldek/rat-ratgdo/blob/main/kicad_files/D1%20Mini%20-%20ESP32/RatGDO-OpenSource-D1Mini-ESP32.pdf)

KnightDoom, 2025-07-08:
Thanks! I was looking at the schematics folder, not the KiCad files.

https://github.com/Kaldek/rat-ratgdo/blob/main/schematics/ratgdo%20open%20source%20D1%20Mini_schem_v6.png

Pmactav, 2025-07-09:
I almost started my own issue for this exact reason, I have a ton of C3 chips laying around but not a single 8266 or compatible ESP32s. Thanks for the ground work, I'll be testing to see if I can get this up and running tomorrow, but likely be back for help! Although its a bit insane to my new guy mind to support a 40 some odd pin WROOM chip but not provide an editable YAML to customize chips. But I'm very new to ESPHOME and its firmware flashing so I'd expect the depth of libraries are the reasoning. 

As a partial answer, the red wire on the Chamberlain devices are sort of 2 way, they both provide power and signal to wall mounted controllers. It seems the RATGDO stops transmission temporarily in some cases as per the readme; either when door action is happening or to temporarily read the position. Likely there was some inconsistency with the transmission and the 1K solves it. See below from readme.

"How ratgdo interfaces with the Chamberlain wiring
The ratgdo taps into the red control wire and uses this single wire for transmission of commands onto the serial bus, and for receiving data from the serial bus. As it is a single wire protocol and operates half-duplex, the ratgdo circuitry uses N-channel MOSFETs wired in a manner which allows the ESP8266 to never transmit when the serial line is already low.

In operation, this means that the ratgdo is mostly listening to the serial bus to ensure it is keeping track of the state of the door. For example, if the door is manually opened (or opened by use of the MyQ app even), these events are broadcast onto the serial bus by the door controller. ratgdo is therefore able to know that the state of the door has changed. In turn, this means that when controlling ratgdo via Home Assistant, the status of the door in HA is always up to date."

geodic, 2025-07-09:
Aah, how could I have missed that? Good catch @Pmactav .

geodic, 2025-09-05:
I flashed the firmware by creating an esphome YAML configuration file, building it, and uploading it. As I mentioned in my first three comments, the C3 needs a special fork of the ratgdo repository to work properly due to the obstruction sensor logic. Flashing instructions are available at [my fork](https://github.com/geodic/ratgdo-esphome) of the ratgdo-esphome repo.

geodic, 2025-11-20:
Hmmm... Can you elaborate a little bit more on the error you are receiving?

Guy559, 2026-02-04:
@geodic I visited your fork but I'm a bit lost.

Where is your modified yaml to use the SuperMini C3? All I see are the same/similar files as pulled from ESPhome-ratgdo but no file to delineate code for ESP32 C3.

Thanks! I'm going to try with a C6...but really want to get your code/design working right. Appreciate your time on this.

geodic, 2026-02-04:
Hi @Guy559, I understand how this can be a point of confusion, so let me elaborate on exactly what my fork does.

TL;DR, just use the "regular YAML files" from my fork. They will work with my schematics.

So basically, as you might (or might not) know, recently, the creator of the original ratgdo completely revamped his product lineup and introduced the new "esp32disco" model based on an ESP32 MCU. However, on the circuit for this board, there is a logic inverter circuit on the obstruction sensor line (I still don't exactly know why), but on my schematic there is not. The way the original code accounts for this is it checks if it is being built for an ESP32 target and if so, automatically inverts the obstruction logic, without checking for anything else. Since my schematics don't have the logic inverter, my fork just removes this logic inversion on the obstruction so that it works normally. The catch is that my fork directly patches the original code responsible for this build target checking, therefore you don't need any special YAML file to enable this patch, you can just use the "regular ones" in the root of my fork.

Guy559, 2026-02-04:
Ah, thank you. I’ll take a look at snagging some C3s. Appreciate the quick responses.

Guy559, 2026-02-04:
Still poking around. This was in discussions tab. Seems someone made it work but using different repository. Not sure how functionality corresponds (or differs) from ESPHone-RATGDO channel.

[https://github.com/Kaldek/rat-ratgdo/discussions/133](url)

Edit: Not sure why link isn't working, but here's the relevant title: "**rat-ratgdo breadboard using XIAO ESP32C6**"

RebbePod, 2026-08-25:
anyone successfully got this to work?

I'm having trouble figuring out the proper schematic for the ESP32-C3

I'm getting this error
```
AttributeError: module 'esphome.components.lock' has no attribute 'LOCK_SCHEMA'
Failed config
Platform not found: 'lock.ratgdo'
```